pi-top/pi-top-Python-SDK #69: Add use cases to "getting started" page

jcapona, 2020-11-17:
In Getting Started, add new sections along the lines of the following use cases:
- Using the pi-top itself (e.g. buttons, OLED, reading battery, etc.)
- Doing physical computing with pi-top [4] (PMA)
- Using peripherals (e.g. pi-topPULSE/pi-topPROTO+)
- Making use of additional helper classes (e.g. keyboard, speech)

Use cases should have at least one associated recipe.

olivierwilkinson, 2020-12-10:
This seems to be partially handled by https://github.com/pi-top/pi-top-Python-SDK/pull/126. Should I hold off until that PR is is resolved until I begin work on this issue?

In the PR above the "Doing physical computing with pi-top [4]" subheading has a good amount of content in it. Can we flesh out this ticket to determine the information we want to present in each subheading? Originally I was going to just add a list of recipes for each.

jcapona, 2020-12-14:
@olivierwilkinson I'll take this ticket over 👍 since I've partially addressed it on the PR you mentioned

PR #126 (2020-12-15): Add overview docs; refactor PMA docs
Closes #68
Closes #69

### Summary

- Reorganize PMA docs
- Fix readthedocs builds
- Add overview page
closed 2020-12-15 by m-roberts