pi-top/pi-top-Python-SDK #127: OLED: handle draw actions while displaying in background thread

m-roberts, 2020-12-08:
Currently, the OLED can play animated images in a background thread. While this is occurring, it is possible to call other functions in the main thread that draw to the display.

This situation should be handled in a more elegant way.

Possible solutions:
* all draw actions are ignored until the animation is explicitly stopped or finished
* draw actions block, background animation ends on next loop
* background action is immediately stopped in favour of next requested action

duwudi, 2021-01-26:
@m-roberts I like the idea of blocking behaviour to wait until current animation loop finishes (option 2) - this is nice to have in simple user programs as you don't have to manage it yourself by waiting with a sleep

Could we add an optional parameter in for `force=True` so that it could force the background action to stop immediately?

m-roberts, 2021-01-26:
The issues with blocking on current animation loop is that it could be a really long GIF, and you would get locked out. Trying to get any smarter than that requires handling timeouts, etc. which just gets messy fast.

I would say that if you provide a newer instruction to display on the OLED, it should either override anything in the background immediately, or fail until the user has explicitly taken action to end what they are doing in the background thread..

m-roberts, 2021-03-26:
Jorge and I agree that newer requests should override

PR #287 (2021-03-26): Stop previously running background thread when displaying anything
Closes #127
Also improves accuracy of FPS regulator by stopping it closer to the display function
closed 2021-03-26 by m-roberts