pi-top/pi-top-Python-SDK #201: Writing text on the OLED screen using the new system

tykavanaugh, 2021-02-02:
I'm not really sure where I'm actually supposed to be looking since the documentation in the pitop OS and the website seems to point in all sorts of directions. I think this is the most up to date pitop repo, though I may be wrong.

I started trying to re-do an previous project that was on the now depreciated codebase. Just trying to basically copy the tutorial has already been quite difficult.

```
import pitop.miniscreen as screen
oled = screen.Miniscreen()

while True:
    oled.display_multiline_text('test')
    sleep(10)
```
Also tried:
```

import pitop.miniscreen
screen = pitop.miniscreen.Miniscreen()
```
Stuff like that, always the same error, which doesn't really tell me whats actually wrong. I'm not the most savvy on OOP, but I don't understand why "init" isn't working inside the miniscreen object. All the tutorials import the object without any arguments, so why does it demand arguments? Looking at the actual code it doesn't make any more sense to me.
> 
>   File "/home/pi/Documents/file.py", line 1, in <module>
>     import pitop.miniscreen as screen
>   File "/usr/lib/python3/dist-packages/pitop/__init__.py", line 1, in <module>
>     from .system.pitop import PiTop
>   File "/usr/lib/python3/dist-packages/pitop/system/pitop.py", line 2, in <module>
>     from pitop.miniscreen import Miniscreen
>   File "/usr/lib/python3/dist-packages/pitop/miniscreen/__init__.py", line 5, in <module>
>     from .buttons import (
>   File "/usr/lib/python3/dist-packages/pitop/miniscreen/buttons/__init__.py", line 2, in <module>
>     from .buttons import (
>   File "/usr/lib/python3/dist-packages/pitop/miniscreen/buttons/buttons.py", line 21, in <module>
>     class Buttons(metaclass=Singleton):
> TypeError: __init__() takes 2 positional arguments but 4 were give

jcapona, 2021-02-02:
hi @tykavanaugh ! 

the 2 snippets you included work for me (just had to import `sleep`), but your traceback makes me think that some of the files in the folder `/usr/lib/python3/dist-packages/pitop/`, where this SDK is installed, might have been modified. I say this because the `init.py` files referenced in your traceback should be named `__init__.py`. 

Can you please try to re-install this package? You can do this in a terminal, running `sudo apt install python3-pitop python3-pitop-full --reinstall`

m-roberts, 2021-02-02:
I can confirm that the snippets work for me (with `sleep` imported).
I am not sure why it would fail like this, but it does indicate that something under-the-hood is not versioned correctly or has been modified directly.

m-roberts, 2021-02-03:
Hi - I am closing this as it doesn't appear to be a problem that we can replicate. If you have any further issues, please consider using the [pi-top forum](https://forum.pi-top.com/) - thanks!

closed 2021-02-03 by m-roberts

tykavanaugh, 2021-02-04:
I figured out the issue, it was actually much more simple. I either hadn't restarted since the last time I ran apt upgrade, or only ran apt update and thought I had run both successfully. Same code worked perfectly fine upon reboot.