pi-top/pi-top-Python-SDK #206: OLED on Raspbian continually flashes "pi-top"

KrazyKerbalnaut, 2021-02-04:
OLED continually flashes "pi-top" logo when using the Raspbian port of the driver software. My concern, as others have stated is OLED burn-in. Myself and others are likely to want to use their own OS'es for the pi as Sirius OS is not quite ready for everyday use cases. It is slow and bulky when compared to Raspbian. For example, VSCode takes as much as 19 seconds to compile one of my 3k lines of code projects when Raspbian can do it in 4-5 seconds.

I will start delving into the driver's for the OLED on both OS'es and see if I can create a pull request with the required information to correct this issue.

m-roberts, 2021-02-04:
Hi - no need! We believe that we have identified the issue and, after testing, will be releasing a fix soon (hopefully today). This issue will be closed once that is done, but in future we would ask that you use https://forum.pi-top.com/ for issues that are not specifically related to the SDK. Thanks!

KrazyKerbalnaut, 2021-02-04:
Apologies, I was using the sdk for something else and I didn't even think to use the forum. I will transfer this and the other bug over to the forum.

Thanks!

closed 2021-02-04 by KrazyKerbalnaut