pi-top/pi-top-Python-SDK #326: Provide simple interface for selecting camera resolution profile

m-roberts, 2021-04-13:
Something like this:
```
camera = Camera(resolution="720p")
```

For these values:
* `720p`
* `480p`
* `360p`
* `240p`

But without strings. Discuss.

duwudi, 2021-06-11:
I don't think there's a better way to do this than using a string, the fact the names start with a number makes any other method difficult. Other options:

- Use names like HD, SD, QVGA etc -  cumbersome and non-specific, they have a one-to-many relationship with actual resolutions
- Large/medium/small/smallest - simple but lacks direct relationship, completely ambiguous as to what resolution is actually being used.

I also don't think it's a good idea to use a string like "720p" either as even those are one-to-many: 720p can mean 1280×720, 1360×768 and 366×768 according to Wikipedia. I'd prefer to stick to the tuple method so that users know exactly what the width and height of the image are - they generally need this information even when doing simple challenges on Further.

duwudi, 2021-06-11:
Instead, we should just add a check to make sure that a valid resolution has been passed, and if it is invalid provide clear instructions on what resolutions they can use.