pi-top/pi-top-Python-SDK #374: Web Controller: Add support for using remote device features

m-roberts, 2021-06-11:
e.g. if accessed from a smartphone, allow support for using the onboard camera/microphone/accelerometer.

m-roberts, 2021-06-11:
So after chatting with @olivierwilkinson, we have determined that [Chrome's security policy will only allow you to access your device's microphone/camera when the site's origin is on https or is called localhost](https://medium.com/@Carmichaelize/enabling-the-microphone-camera-in-chrome-for-local-unsecure-origins-9c90c3149339).

Cloudflare offers a [free tunnelling service](https://blog.cloudflare.com/tunnel-for-everyone/) that will likely be sufficient for our purposes here.

m-roberts, 2021-06-11:
Some ideas that springboard off this -

Multiplayer games! Each client that connects to the robot's server gets their own socket and the broadcast method sends to all of them.

e.g. a 'game' where there's a robot and multiple players trying to make it do different things competing by using their phone's gyros, mics, cameras
e.g. a robot where each person who joins can only control 1 part at a time

duwudi, 2021-06-15:
> So after chatting with @olivierwilkinson, we have determined that [Chrome's security policy will only allow you to access your device's microphone/camera when the site's origin is on https or is [...]

Does this require a unique URL to be set up for each user every time they run it?

duwudi, 2021-06-15:
https://www.chromium.org/Home/chromium-security/deprecating-powerful-features-on-insecure-origins

adding this reference to the original chromium security notice

duwudi, 2021-06-17:
Stumbled on [ngrok](https://ngrok.com/) today and tried tunnelling the web controller through that - it works and I have an https URL but it's painfully slow - I've read that Cloudflare argo tunnel is a bit quicker but I don't think it will be good enough. Unless we can get a TLS connection through a web URL but send all the data over the local network (like Further does) I don't think this will be a viable solution - is this possible?