pi-top/pi-top-Python-SDK #376: Add text-to-speech functionality

duwudi, 2021-06-16:
Use pi-top's onboard speaker to interact with the user via speech - particularly useful/interesting for robotics applications but can also have a lot of value for simpler Foundation Kit projects.

See old library created [here ](https://github.com/pi-top/python3-pt-speech-deb)that uses `espeak`. 

- Possibly make this part of the `Pitop` class so it can be used by default without the user specifically adding it manually
- Would be nice to have multiple implementations available to use, i.e. user can choose between a high-quality cloud-based version (Google/Amazon/Microsoft) or a lower quality local version (espeak etc)

angusjfw, 2021-07-22:
I got pretty good results using festival with a plug-in voice as described [here](https://askubuntu.com/a/908889).

```
sudo apt-get install -y festival festvox-us-slt-hts
festival -b '(voice_cmu_us_slt_arctic_hts)' \
    '(SayText "The temperature is 22 degrees centigrade and there is a slight breeze from the west.")'
```

That python lib actually binds to the c++ lib, subprocesses are only used for converting to mp3 I think. It gives me an ImportError when I try to use it though.

angusjfw, 2021-07-22:
Similar results with flite:
```
sudo apt install flite
flite -voice slt -t "The Raspberry Pi is a great Maker platform!"
```

duwudi, 2021-07-22:
> I got pretty good results using festival with a plug-in voice as described [here](https://askubuntu.com/a/908889).

This one sounds pretty good actually, let's just use that!

duwudi, 2021-07-22:
> The [python lib](https://github.com/techiaith/pyfestival) actually binds to the c++ lib, subprocesses are only used for converting to mp3 I think. It gives me an ImportError when I try to use it though.

Can we still use the plugin if using the python library?

duwudi, 2021-07-28:
> It gives me an ImportError when I try to use it though.

Yeah I've got the same thing, the pypy package is not actually up to date with the repo though - importing from the pip-installed version gives this error:

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/festival.py", line 3, in <module>
    from . import _festival
ImportError: attempted relative import with no known parent package
```

Installing from the repo directly gives this error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/festival.py", line 6, in <module>
    import _festival
ImportError: /lib/arm-linux-gnueabihf/libestools.so.2.5: undefined symbol: omp_get_thread_num
```

Not sure how to fix this, any ideas?

duwudi, 2021-07-28:
> Didn't try the [python bindings](https://github.com/happyalu/pyflite)

Can't get this one building either, fails on `python3 setup.py build` even after installing `swig`

duwudi, 2021-07-28:
> ```
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/pi/.local/lib/python3.7/site-packages/festival.py", line 6, in <module>
>     import _festival
> ImportError: /lib/arm-linux-gnueabihf/libestools.so.2.5: undefined symbol: omp_get_thread_num
> ```

Adding this fix https://github.com/techiaith/pyfestival/pull/9 gets the python bindings working. The plan is to fork this repo and manage it ourselves since it's relatively simple