pi-top/pi-top-Python-SDK #604: No IMU in PMA?

miquelmassot, 2023-02-16:
### Summary
I've got a Pi-Top [4] with the expansion plate and robotic kit. If I run
```python
from pitop.pma.imu import IMU
imu = IMU()
```

I get:
 
```
Exception: This Expansion Plate doesn't have an IMU
```

*Is this supposed to happen? According to the dealer and the documentation, it does have an IMU.*
[Reference](https://uk.rs-online.com/web/p/raspberry-pi-hats-add-ons/2162810)

This is the output of pi-top devices

```
pi@pi-top:~ $ pi-top devices
HUB ======================================================================================================================================================================================
pi-top [4] (v5.4)
PERIPHERALS ==============================================================================================================================================================================
[   ] pi-top [4] Foundation Plate 
[ ✓ ] pi-top [4] Expansion Plate (v22.1)
[   ] pi-top Touchscreen 
[   ] pi-top Keyboard 
[   ] pi-topPULSE 
[   ] pi-topSPEAKER (v1) - Left channel 
[   ] pi-topSPEAKER (v1) - Mono 
[   ] pi-topSPEAKER (v1) - Right channel 
[   ] pi-topSPEAKER (v2) 
[   ] pi-topPROTO+ 

```

### Environment Information

```bash
pi@pi-top:~ $ uname -a
Linux pi-top 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux
pi@pi-top:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
pi@pi-top:~ $ python --version
Python 3.9.2
pi@pi-top:~ $ dpkg --list | grep pitop
ii  python3-pitop                        0.30.0-1                             all          pi-top Python 3 Library - Main
ii  python3-pitop-battery                0.30.0-1                             all          pi-top Python 3 Library - Battery
ii  python3-pitop-camera                 0.30.0-1                             all          pi-top Python 3 Library - Camera
ii  python3-pitop-cli                    0.30.0-1                             all          pi-top Python 3 Library - CLI
ii  python3-pitop-common                 0.30.0-1                             all          pi-top Python 3 Library - Common
ii  python3-pitop-core                   0.30.0-1                             all          pi-top Python 3 Library - Core
ii  python3-pitop-display                0.30.0-1                             all          pi-top Python 3 Library - Display
ii  python3-pitop-doc                    0.30.0-1                             all          pi-top Python 3 Library - Documentation and Examples
ii  python3-pitop-full                   0.30.0-1                             all          pi-top Python 3 Library - Complete
ii  python3-pitop-keyboard               0.30.0-1                             all          pi-top Python 3 Library - Keyboard
ii  python3-pitop-miniscreen             0.30.0-1                             all          pi-top Python 3 Library - Main
ii  python3-pitop-pma                    0.30.0-1                             all          pi-top Python 3 Library - PMA
ii  python3-pitop-processing             0.30.0-1                             all          pi-top Python 3 Library - Processing
ii  python3-pitop-robotics               0.30.0-1                             all          pi-top Python 3 Library - Robotics
ii  python3-pitop-simulation             0.30.0-1                             all          pi-top Python 3 Library - System
ii  python3-pitop-system                 0.30.0-1                             all          pi-top Python 3 Library - System

```

miquelmassot, 2023-02-16:
This is what I hit:

https://github.com/pi-top/pi-top-Python-SDK/blob/13cf6d63e00a92024fc4306e62bedd356693332b/packages/pma/pitop/pma/imu_controller.py#L30

jcapona, 2023-02-16:
Hi @miquelmassot; the older Expansion Plates had an IMU in it but the latest version (v4) doesn't. It seems that your unit is one of the latest; you can learn the hardware version of your Expansion Plate by running this code in a python interpreter:

```
from pitop.common.firmware_device import FirmwareDevice, FirmwareDeviceID
# make sure the Expansion Plate is attached to the pi-top
device = FirmwareDevice(FirmwareDeviceID.pt4_expansion_plate)
device.get_sch_hardware_version_major()
```

miquelmassot, 2023-02-17:
Thanks for the clarification @jcapona.

This information does not show up in any manual, specsheet or datasheet. I bought the product believing its got the features stated in the product page.

I don't think the retailer will let me return the kits that are already opened. This issue is costing me time and effort in getting these kits up and running for teaching at the university.

Can I replace my v4 expansion boards with the older ones?

miquelmassot, 2023-02-17:
[image]

jcapona, 2023-02-21:
@miquelmassot I've raised this issue with our customer support and they should have reached out to you by email to resolve it

closed 2023-02-21 by jcapona

miquelmassot, 2023-02-21:
Thanks @jcapona, I'm in touch with support at the moment.