pi-top/pi-top-Python-SDK #659: Having hard time with camera

vovikdrg, 2024-08-16:
My Kids are doing line following but they are constantly getting an error

Exception in thread Thread-7:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/pitop/camera/camera.py", line 294, in __process_camera_output
    self._frame_handler.frame = self._camera.get_frame()
  File "/usr/lib/python3/dist-packages/pitop/camera/core/cameras/usb_camera.py", line 75, in get_frame
    raise ValueError("Couldn't grab frame from camera")
ValueError: Couldn't grab frame from camera

They do have a loop where they try to grab cam.get_frame and this error occurs.

jcapona, 2024-08-16:
hi @vovikdrg !

We've seen this issue happen sometimes; you can try:
- Connecting the USB camera into another USB port; try connecting it into the Expansion Plate's USB port first.
- Replacing the camera's USB cable.

If you're comfortable using a terminal, you can ssh into the pi-top and check the journal when you're plugging the cameras (running `journalctl -f`) and check if there's some information in there...

vovikdrg, 2024-08-18:
Thanks for that. Now as they connect camera to expansion plate its kind of okay. But i feel camera quality is bad so ordered new one.

vovikdrg, 2024-08-19:
Another lesson today, seems to be connecting to expansion plate usb make camera stable, but we stopped using camera, and using OpenCV directly + try catch to release  camera. Not sure if real issue can be that somehow sdk does not release camera on exit